sparkfun/SparkFun_LSM6DS3_Arduino_Library #6: Compilation error with ESP32 MSBFIRST

ThingEngineer, 2018-10-18:
Compilation on on the ESP32 results in this error:
'MSBFIRST' was not declared in this scope

<Arduino.h> must be included into SparkFunLSM6DS3.cpp

Imara90, 2018-12-11:
This issue might just be a linking problem, but I'm having the same problem. I get an error saying there is no such file as Wire.h, and when trying to include <Arduino.h> it actually does the same. I added the lib folder of the LSM to the compiler.mk file and I have the arduino esp32 component in the project. Any idea on what may be wrong here?

edspark, 2019-03-12:
Hey, thank you guys for filing an issue but in particular thank you @ThingEngineer for the pull request.  The issue was that the library sets the SPI bit order using  Arduino's MSBFIRST define, when ESP32 uses the define: SPI_MSBFIRST. I've added a check in the library for ESP32 to fix the issue. I've also included Arduino.h rather than just stdint.h. Changes can be found here: https://github.com/sparkfun/SparkFun_LSM6DS3_Arduino_Library/commit/9eb64645716cd161ec8019648a7e572c0b32202d. 

Thanks again for filing an issue and sorry for the late reply.

closed 2019-03-12 by edspark