sparkfun/SparkFun_LSM6DS3_Arduino_Library #17: Examples, ODR bit code does nothing

lienbacher, 2020-06-21:
I noticed in the examples code that the [2 lines that set the ODR bit in CTRL4_C register](https://github.com/sparkfun/SparkFun_LSM6DS3_Arduino_Library/blob/88488f9d08e1a7385e8031ff8d21605094006072/examples/InterruptFreeFallConfig/InterruptFreeFallConfig.ino#L79) do nothing:

`//Set the ODR bit
  errorAccumulator += myIMU.readRegister(&dataToWrite, LSM6DS3_ACC_GYRO_CTRL4_C);
  dataToWrite &= ~((uint8_t)LSM6DS3_ACC_GYRO_BW_SCAL_ODR_ENABLED);`

dataToWrite is overwritten with an empty register. In the next line - if I am not mistaken, because I may be - the bit is actually unset from the register. Then nothing is done with dataToWrite code. Certainly there is not a single bit changed in any register in this code, although reading the Datasheet and lining it up with what the rest of the code does it certainly seems that the intention of the code is to set Bit 8 of CTRL4_C register to 1 / register 0x13 to 0x80.

edspark, 2021-11-09:
I'm trying to catch up but I'm unsure if I'm even looking at the correct datasheet. I know it's been a while since you posted this issue but hopefully you still have a little bit of interest. Looking at the code, `dataToWrite` is used initially to set the bandwidth, force, and output data rate and is [indeed written](https://github.com/sparkfun/SparkFun_LSM6DS3_Arduino_Library/blob/d838d4eb36593b7f26d65850052c66a681937531/examples/InterruptFreeFallConfig/InterruptFreeFallConfig.ino#L71-L76). However yes in the very next line the ODR bit is brought into dataToWrite, but then is never written to a register. Unfortunately here's where I was stopped short because the datasheet I found did not align with the [notes](https://github.com/sparkfun/SparkFun_LSM6DS3_Arduino_Library/blob/d838d4eb36593b7f26d65850052c66a681937531/src/SparkFunLSM6DS3.h#L1022-L1031) for the `LSM6DS3_ACC_GYRO_BW_SCAL_ODR_ENABLED` variable. I did test free fall and it works for me, so I'm not sure what the original intent for setting that bit is.

edspark, 2021-11-09:
I think I'll close this stale issue in the mean time. Clearly setting that bit does nothing for the sketch and so I [removed it](https://github.com/sparkfun/SparkFun_LSM6DS3_Arduino_Library/commit/e881b23266541e7625370581a1671dcdefdb0017).

closed 2021-11-09 by edspark